wasm-micro-runtime/wasm-micro-runtime #4277: WAMR 2.3.0 fails to build on armv7 and riscv64 (Alpine Linux, musl libc)

jirutka, 2025-05-14:
Build flags: https://gitlab.alpinelinux.org/alpine/aports/-/blob/ff7271de8e24204c82226a993efe4d7ed777b073/testing/wasm-micro-runtime/APKBUILD#L60-84

```
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6104:45: warning: value computed is not used [-Wunused-value]
 6104 | #define SIMD_LANE_HANDLE_UNALIGNED_ACCESS() *frame_ip++;
      |                                             ^~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6109:9: note: in expansion of macro 'SIMD_LANE_HANDLE_UNALIGNED_ACCESS'
 6109 |         SIMD_LANE_HANDLE_UNALIGNED_ACCESS();                   \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6125:25: note: in expansion of macro 'SIMD_EXTRACT_LANE_OP'
 6125 |                         SIMD_EXTRACT_LANE_OP(i8x16, int8, PUSH_I32);
      |                         ^~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6104:45: warning: value computed is not used [-Wunused-value]
 6104 | #define SIMD_LANE_HANDLE_UNALIGNED_ACCESS() *frame_ip++;
      |                                             ^~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6109:9: note: in expansion of macro 'SIMD_LANE_HANDLE_UNALIGNED_ACCESS'
 6109 |         SIMD_LANE_HANDLE_UNALIGNED_ACCESS();                   \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6130:25: note: in expansion of macro 'SIMD_EXTRACT_LANE_OP'
 6130 |                         SIMD_EXTRACT_LANE_OP(i8x16, uint8, PUSH_I32);
      |                         ^~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6104:45: warning: value computed is not used [-Wunused-value]
 6104 | #define SIMD_LANE_HANDLE_UNALIGNED_ACCESS() *frame_ip++;
      |                                             ^~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6116:9: note: in expansion of macro 'SIMD_LANE_HANDLE_UNALIGNED_ACCESS'
 6116 |         SIMD_LANE_HANDLE_UNALIGNED_ACCESS();                  \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6135:25: note: in expansion of macro 'SIMD_REPLACE_LANE_OP'
 6135 |                         SIMD_REPLACE_LANE_OP(i8x16, int8, POP_I32);
      |                         ^~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6104:45: warning: value computed is not used [-Wunused-value]
 6104 | #define SIMD_LANE_HANDLE_UNALIGNED_ACCESS() *frame_ip++;
      |                                             ^~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6109:9: note: in expansion of macro 'SIMD_LANE_HANDLE_UNALIGNED_ACCESS'
 6109 |         SIMD_LANE_HANDLE_UNALIGNED_ACCESS();                   \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6140:25: note: in expansion of macro 'SIMD_EXTRACT_LANE_OP'
 6140 |                         SIMD_EXTRACT_LANE_OP(i16x8, int16, PUSH_I32);
      |                         ^~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6104:45: warning: value computed is not used [-Wunused-value]
 6104 | #define SIMD_LANE_HANDLE_UNALIGNED_ACCESS() *frame_ip++;
      |                                             ^~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6109:9: note: in expansion of macro 'SIMD_LANE_HANDLE_UNALIGNED_ACCESS'
 6109 |         SIMD_LANE_HANDLE_UNALIGNED_ACCESS();                   \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6145:25: note: in expansion of macro 'SIMD_EXTRACT_LANE_OP'
 6145 |                         SIMD_EXTRACT_LANE_OP(i16x8, uint16, PUSH_I32);
      |                         ^~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6104:45: warning: value computed is not used [-Wunused-value]
 6104 | #define SIMD_LANE_HANDLE_UNALIGNED_ACCESS() *frame_ip++;
      |                                             ^~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6116:9: note: in expansion of macro 'SIMD_LANE_HANDLE_UNALIGNED_ACCESS'
 6116 |         SIMD_LANE_HANDLE_UNALIGNED_ACCESS();                  \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6150:25: note: in expansion of macro 'SIMD_REPLACE_LANE_OP'
 6150 |                         SIMD_REPLACE_LANE_OP(i16x8, int16, POP_I32);
      |                         ^~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6104:45: warning: value computed is not used [-Wunused-value]
 6104 | #define SIMD_LANE_HANDLE_UNALIGNED_ACCESS() *frame_ip++;
      |                                             ^~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6109:9: note: in expansion of macro 'SIMD_LANE_HANDLE_UNALIGNED_ACCESS'
 6109 |         SIMD_LANE_HANDLE_UNALIGNED_ACCESS();                   \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6155:25: note: in expansion of macro 'SIMD_EXTRACT_LANE_OP'
 6155 |                         SIMD_EXTRACT_LANE_OP(i32x4, int32, PUSH_I32);
      |                         ^~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6104:45: warning: value computed is not used [-Wunused-value]
 6104 | #define SIMD_LANE_HANDLE_UNALIGNED_ACCESS() *frame_ip++;
      |                                             ^~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6116:9: note: in expansion of macro 'SIMD_LANE_HANDLE_UNALIGNED_ACCESS'
[... 193 lines cut ...]
      |                                                    uint8 * {aka unsigned char *}
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6608:9: note: in expansion of macro 'SIMD_LOAD_LANE_COMMON'
 6608 |         SIMD_LOAD_LANE_COMMON(vec, register, lane, width); \
      |         ^~~~~~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6613:25: note: in expansion of macro 'SIMD_LOAD_ZERO_OP'
 6613 |                         SIMD_LOAD_ZERO_OP(i32x4, 32);
      |                         ^~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/../common/wasm_runtime_common.h:160:27: note: expected 'uint32 *' {aka 'unsigned int *'} but argument is of type 'uint8 *' {aka 'unsigned char *'}
  160 | GET_I64_FROM_ADDR(uint32 *addr)
      |                   ~~~~~~~~^~~~
[... cut ...]
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/wasm_interp_fast.c:6618:25: note: in expansion of macro 'SIMD_LOAD_ZERO_OP'
 6618 |                         SIMD_LOAD_ZERO_OP(i64x2, 64);
      |                         ^~~~~~~~~~~~~~~~~
/builds/alpine/aports/testing/wasm-micro-runtime/src/wasm-micro-runtime-WAMR-2.3.0/core/iwasm/interpreter/../common/wasm_runtime_common.h:160:27: note: expected 'uint32 *' {aka 'unsigned int *'} but argument is of type 'uint8 *' {aka 'unsigned char *'}
  160 | GET_I64_FROM_ADDR(uint32 *addr)
      |  
```

lum1n0us, 2025-05-15:
Thank you for reporting it. I'm just curious, why does `-incompatible-pointer-type` trigger an error in your environment but only a warning in mine? My first guess is that you're using `-Werror=incompatible-pointer-types`, but I didn't find it in your build flags. Please help me by answering this.

My ultimate goal is to set a similar rule to prevent regression.

jirutka, 2025-05-18:
Tbh, I don’t know. abuild sets the following flags by default: `-Os -fstack-clash-protection -Wformat -Werror=format-security`. There are also some defaults backed in gcc. Basically, every distro has slightly different defaults.

closed 2025-05-19 by lum1n0us