wasm-micro-runtime/wasm-micro-runtime #4280: When using msvc to compile the sample in samples\basic, there is an error: "definition of dllimport function not allowed"

UradaSources, 2025-05-16:
When using msvc to compile the sample in samples\basic, there is an error: "definition of dllimport function not allowed"

i am trying to compile the samples\basic sample program in Windows using the Visual Studio CMake folder mode, but the compilation always fails, and the compiler reports C2491 errors, which seem to be caused by the macro:
`#define WASM_RUNTIME_API_EXTERN __declspec(dllimport)`

the location is 2 functions under the core\iwasm\common\wasm_runtime_common.c file
```
WASM_RUNTIME_API_EXTERN bool
wasm_runtime_resolve_symbols(WASMModuleCommon *module)

WASM_RUNTIME_API_EXTERN bool
wasm_runtime_is_underlying_binary_freeable(WASMModuleCommon *const module)
```

There are also a lot of C4273 dll link inconsistency warnings. Did I do something wrong?

Also, does the project officially support clang under Windows? I tried to compile the project with mingw-clang, and there were always a lot of strange errors

```
[proc] 执行命令: "C:\Program Files\CMake\bin\cmake.EXE" --build d:/lch/Private/wasm_test/wasm-micro-runtime/samples/basic/build --config Debug --target all --
[build] [10/88   1% :: 1.886] Building C object CMakeFiles/vmlib.dir/D_/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_socket.c.obj
[build] In file included from D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_socket.c:6:
[build] In file included from D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/../include/platform_api_vmcore.h:9:
[build] In file included from D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/../include/platform_common.h:13:
[build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/platform_internal.h:14:9: warning: unknown pragma ignored [-Wunknown-pragmas]
[build]    14 | #pragma warning(disable : 5105)
[build]       |         ^
[build] In file included from D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_socket.c:9:
[build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_util.h:15:9: warning: unknown pragma ignored [-Wunknown-pragmas]
[build]    15 | #pragma warning(disable : 5105)
[build]       |         ^
[build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_socket.c:12:9: warning: unknown pragma ignored [-Wunknown-pragmas]
[build]    12 | #pragma comment(lib, "ws2_32.lib")
[build]       |         ^
[build] 3 warnings generated.
[build] [11/88   2% :: 1.934] Building C object CMakeFiles/vmlib.dir/D_/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c.obj
[build] FAILED: CMakeFiles/vmlib.dir/D_/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c.obj 
[build] C:\msys64\clang64\bin\clang.exe -DBH_DEBUG=1 -DBH_FREE=wasm_runtime_free -DBH_MALLOC=wasm_runtime_malloc -DBH_PLATFORM_WINDOWS -DBUILD_TARGET_X86_64 -DHAVE_STRUCT_TIMESPEC -DWAMR_ENABLE_COPY_CALLSTACK=0 -DWASM_DISABLE_HW_BOUND_CHECK=0 -DWASM_DISABLE_STACK_HW_BOUND_CHECK=0 -DWASM_DISABLE_WAKEUP_BLOCKING_OP=0 -DWASM_ENABLE_AOT=1 -DWASM_ENABLE_AOT_INTRINSICS=1 -DWASM_ENABLE_BULK_MEMORY=1 -DWASM_ENABLE_FAST_INTERP=0 -DWASM_ENABLE_INTERP=1 -DWASM_ENABLE_LIBC_BUILTIN=1 -DWASM_ENABLE_LIBC_WASI=1 -DWASM_ENABLE_MINI_LOADER=0 -DWASM_ENABLE_MODULE_INST_CONTEXT=1 -DWASM_ENABLE_MULTI_MODULE=0 -DWASM_ENABLE_QUICK_AOT_ENTRY=1 -DWASM_ENABLE_SHARED_MEMORY=0 -DWASM_ENABLE_SHRUNK_MEMORY=1 -DWASM_GLOBAL_HEAP_SIZE=10485760 -DZYCORE_STATIC_BUILD=1 -DZYDIS_STATIC_BUILD=1 -D_WINSOCK_DEPRECATED_NO_WARNINGS -ID:/lch/Private/wasm_test/wasm-micro-runtime/core/iwasm/interpreter -ID:/lch/Private/wasm_test/wasm-micro-runtime/core/iwasm/aot -ID:/lch/Private/wasm_test/wasm-micro-runtime/samples/basic/build/_deps/zycore-src/include -ID:/lch/Private/wasm_test/wasm-micro-runtime/samples/basic/build/_deps/zycore-build -ID:/lch/Private/wasm_test/wasm-micro-runtime/samples/basic/build/_deps/zydis-build -ID:/lch/Private/wasm_test/wasm-micro-runtime/samples/basic/build/_deps/zydis-src/include -ID:/lch/Private/wasm_test/wasm-micro-runtime/samples/basic/build/_deps/zydis-src/src -ID:/lch/Private/wasm_test/wasm-micro-runtime/core/iwasm/libraries/libc-builtin -ID:/lch/Private/wasm_test/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/include -ID:/lch/Private/wasm_test/wasm-micro-runtime/core/iwasm/libraries/libc-wasi/sandboxed-system-primitives/src -ID:/lch/Private/wasm_test/wasm-micro-runtime/samples/basic/../../core/iwasm/include -ID:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows -ID:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/../include -ID:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/common/libc-util -ID:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/mem-alloc -ID:/lch/Private/wasm_test/wasm-micro-runtime/core/iwasm/common -ID:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/utils -ID:/lch/Private/wasm_test/wasm-micro-runtime/samples/basic/src -ID:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/utils/uncommon -Wall -Wextra -Wformat -Wformat-security -std=gnu99 -ffunction-sections -fdata-sections                                          -Wall -Wno-unused-parameter -Wno-pedantic -g -MD -MT CMakeFiles/vmlib.dir/D_/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c.obj -MF CMakeFiles\vmlib.dir\D_\lch\Private\wasm_test\wasm-micro-runtime\core\shared\platform\windows\win_file.c.obj.d -o CMakeFiles/vmlib.dir/D_/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c.obj -c D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c
[build] In file included from D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c:6:
[build] In file included from D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/../include/platform_api_extension.h:9:
[build] In file included from D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/../include/platform_common.h:13:
[build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/platform_internal.h:14:9: warning: unknown pragma ignored [-Wunknown-pragmas]
[build]    14 | #pragma warning(disable : 5105)
[build]       |         ^
[build] In file included from D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c:8:
[build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_util.h:15:9: warning: unknown pragma ignored [-Wunknown-pragmas]
[build]    15 | #pragma warning(disable : 5105)
[build]       |         ^
[build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c:12:9: warning: unknown pragma ignored [-Wunknown-pragmas]
[build]    12 | #pragma comment(lib, "Pathcch.lib")
[build]       |         ^
[build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c:87:13: warning: label followed by a declaration is a C23 extension [-Wc23-extensions]
[build]    87 |             FILE_ATTRIBUTE_TAG_INFO file_info;
[build]       |             ^
[build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c:865:13: warning: label followed by a declaration is a C23 extension [-Wc23-extensions]
[build]   865 |             bool success = CloseHandle(handle->raw.handle);
[build]       |             ^
[build] C:/msys64/clang64/lib/clang/20/include/stdbool.h:24:14: note: expanded from macro 'bool'
[build]    24 | #define bool _Bool
[build]       |              ^
[build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c:872:13: warning: label followed by a declaration is a C23 extension [-Wc23-extensions]
[build]   872 |             int ret = closesocket(handle->raw.socket);
[build]       |             ^
[build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c:1298:31: warning: backslash and newline separated by space [-Wbackslash-newline-escape]
[build]  1298 |             // Starts with \??\ 
[build]       |                               ^
[build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c:1300:20: error: expected identifier
[build]  1300 |                 && ((wbuf[4] >= L'A' && wbuf[4] <= L'Z')
[build]       |                    ^
[... cut ...]
[build] In file included from D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/../include/platform_common.h:13:
[build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/platform_internal.h:14:9: warning: unknown pragma ignored [-Wunknown-pragmas]
[build]    14 | #pragma warning(disable : 5105)
[build]       |         ^
[build] 1 warning generated.
[build] [11/88  12% :: 2.304] Building CXX object CMakeFiles/vmlib.dir/D_/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_atomic.cpp.obj
[build] ninja: build stopped: subcommand failed.
```

lum1n0us, 2025-05-19:
Just want to make sure that we are on the same page. Based on the log above, the error is here, which is not the same as the content of this issue. Could you help us confirm that?

```shell
[build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c:1300:20: error: expected identifier
[build]  1300 |                 && ((wbuf[4] >= L'A' && wbuf[4] <= L'Z')
```

UradaSources, 2025-05-20:
> Just want to make sure that we are on the same page. Based on the log above, the error is here, which is not the same as the content of this issue. Could you help us confirm that?
> 
> [build] D:/lch/Private/wasm_test/wasm-micro-runtime/core/shared/platform/windows/win_file.c:1300:20: error: expected identifier
> [build]  1300 |                 && ((wbuf[4] >= L'A' && wbuf[4] <= L'Z')
> I mean if you are using msvc, `CC` should be something like _VC/Tools/MSVC/_/bin/Hostx64/x64/cl.exe*. So, why _mingw-clang_ here?

Oh, I just tried all the windows toolchains i have, first msvc then mingw-clang, and none of them worked. I have now moved on to wsl and I have no problems compiling on linux. I have successfully compiled an example and run the tests

lum1n0us, 2025-05-22:
Thanks for identifying all those problems. Mostly, we rely on community resources to support WAMR on Windows. There are some aspects that aren't covered. We always suggest using Ubuntu as your development environment.

lum1n0us, 2025-05-22:
- [ ] #4286 
- [ ] about C4273
- [ ] about WASM_RUNTIME_API_EXTERN on windows

UradaSources, 2025-05-22:
> Thanks for identifying all those problems. Mostly, we rely on community resources to support WAMR on Windows. There are some aspects that aren't covered. We always suggest using Ubuntu as your development environment.

OK, thanks

closed 2025-05-22 by UradaSources