wasm-micro-runtime/wasm-micro-runtime #4288: distribute headers for our extensions for wasm module developers

yamt, 2025-05-22:
we have a few things which define the abi/api for wasm modules.
including:
```
core/iwasm/libraries/wasi-nn/include/wasi_nn.h
core/iwasm/libraries/lib-socket/inc/wasi_socket_ext.h
```

it would be more user-friendly to distribute those files as a separate package/tarball/whatever
because a developer of wasm module usually doesn't want to have a copy of the whole wamr source tree
just to use those apis.

lum1n0us, 2025-05-23:
[WAMR_SDK](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/wamr-sdk) was intended to be a toolchain that allows users to follow WAMR's private WASM headers. It initially included three features: wasi-nn (very early version), pthread, and sockets. Later, wasi developed its own thread implementation, wasi-thread, and the use of pthread in WAMR is not encouraged. Similarly, wasi-nn is also not encouraged due to the creation of an (almost) [official Rust API](https://github.com/bytecodealliance/wasi-nn).

Undoubtedly, we were considering not maintaining the WAMR_SDK before this issue was raised.

WAMR_SDK was designed to work with `WAMR_BUILD_LIBC_BUILTIN` (and app-framework, which is also not maintained; you might find it in .github/workflows/build_wamr_sdk.yml). However, I guess now we are hoping it's in a form like wasi-libc + socket. Please let me know your comments about WAMR_SDK.

yamt, 2025-06-03:
> [WAMR_SDK](https://github.com/bytecodealliance/wasm-micro-runtime/tree/main/wamr-sdk) was intended to be a toolchain that allows users to follow WAMR's private WASM headers. It initially included [...]

as you said, wamr-sdk has been for a non-wasi different target. (`WAMR_BUILD_LIBC_BUILTIN`, `WAMR_BUILD_LIB_PTHREAD`, etc)
i'm more interested in wasi and our extensions to it.

as wasi already has a nice sdk (wasi-sdk), i guess it's simpler for us to ship a small library, which only contains our extensions.
users can use wasi-sdk to build their app, linking our extension library.
what do you think?

yamt, 2025-06-09:
> as wasi already has a nice sdk (wasi-sdk), i guess it's simpler for us to ship a small library, which only contains our extensions.
> users can use wasi-sdk to build their app, linking our extension library.

something along the line: https://github.com/bytecodealliance/wasm-micro-runtime/pull/4344

lum1n0us, 2025-06-23:
- [ ] #4344
- [ ] #4360
- [ ] #4392
- [ ] #4394

yamt, 2025-08-12:
wasmr-wasi-extensions is now a part of releases.