wasm-micro-runtime/wasm-micro-runtime #4290: public header location incompatibility

yamt, 2025-05-22:
https://github.com/bytecodealliance/wasm-micro-runtime/pull/4076 changed the location of wasm_export.h etc. 

problems:

* it's an api breaking change. that is, `#include <wasm_export.h>` vs `#include <iwasm/wasm_export.h>`
* it has not been mentioned as a breaking change in [2.3.0 release notes](https://github.com/bytecodealliance/wasm-micro-runtime/releases/tag/WAMR-2.3.0).
* it causes inconsistencies among platforms. for example, nuttx-apps still provides these headers [this way](https://github.com/apache/nuttx-apps/blob/a5709532577ff0b3276a78bc8d278e8bf8a1b037/interpreters/wamr/Make.defs#L26-L27).
* changes like this should be done a bit more carefully. eg. provide compat headers for a few releases.
* there seems no CI tests to use wamr via a cmake package.

lum1n0us, 2025-05-23:
I might miss the point of "it's an api breaking change. that is, #include <wasm_export.h> vs #include <iwasm/wasm_export.h>". Would you mind explaining more clearly?

yamt, 2025-05-23:
> I might miss the point of "it's an api breaking change. that is, #include <wasm_export.h> vs #include <iwasm/wasm_export.h>". Would you mind explaining more clearly?

user applications need to be adapted to use `#include <iwasm/wasm_export.h>`.
or they need to add another `-isystem` to their CFLAGS.
or whatever.

yamt, 2025-05-23:
> there seems no CI tests to use wamr via a cmake package.

https://github.com/bytecodealliance/wasm-micro-runtime/pull/4291

yamt, 2025-05-23:
> or they need to add another `-isystem` to their CFLAGS.

either via the cmake package or manually. please note that not every user app uses cmake.

lum1n0us, 2025-05-23:
> `cmake --build build-wamr -t install`

Now, I understand what you are talking about. Using installation is a new approach for me regarding how to integrate WAMR into a project. I used to think people always embedded WAMR source code into their project and directly used headers in the source code tree and libraries in the build directories. So we didn't treat the new installation destination seriously.

Now, the new installation tree is as follows. We're open to rearranging it, so please let me know your suggestions.

```shell
├── bin
│   ├── iwasm -> iwasm-2.2.0
│   └── iwasm-2.2.0
├── include
│   └── iwasm
│       ├── lib_export.h
│       ├── wasm_c_api.h
│       └── wasm_export.h
└── lib
    ├── cmake
    │   └── iwasm
    │       ├── iwasmConfig.cmake
    │       ├── iwasmConfigVersion.cmake
    │       ├── iwasmTargets.cmake
    │       └── iwasmTargets-release.cmake
    └── libiwasm.a 
```

yamt, 2025-05-26:
> Now, the new installation tree is as follows. We're open to rearranging it, so please let me know your suggestions.

my suggestion: https://github.com/bytecodealliance/wasm-micro-runtime/pull/4295

closed 2025-05-28 by lum1n0us