wasm-micro-runtime/wasm-micro-runtime #4292: Polybench benchmark unnatural performance numbers

Lukasdoe, 2025-05-23:
### Subject of the issue

I am baffled with the results I get from the polybench benchmarks. I will try to attach a graph of my results, but basically, in all but 5 of the benchmarks, AOT outperforms native execution, sometimes be more than 3x.

I also checked that they work correctly and that the DCE-blocker actually function, but even when generating the output (using `-DPOLYBENCH_DUMP_ARRAYS`), the AOT version is significantly faster than native (`clang -O3`, clang 20.1.4). Did any of you already encounter this and has a good explanation to why these benchmarks are so much faster than native?

### Test case

The polybench benchmarks, especially `correlation` in `/tests/benchmarks/polybench`.

### Your environment

* Host OS: Ubuntu 24.04.1 LTS x86_64
* WAMR: 2.3.0, AOT

### Steps to reproduce

Build and execute the polybench benchmark suite. Results vary between systems, but on my Intel i9-7900X (20) @ 4.500GHz, they have been most pronounced.

### Expected behavior

AOT should still be slower than native execution.

### Actual behavior

AOT is significantly faster than native execution.

Lukasdoe, 2025-05-24:
[image]

Lukasdoe, 2025-05-24:
[image]

Septa2112, 2025-05-28:
For the `correlation` case, I used `objdump` to look at the machine code from both the native and AOT builds. It turns out the AOT version has some SIMD instructions that the native one doesn’t, like `vucomisd`, `vmulsd`, and `vcvtusi2sd`. These might be part of the reason why AOT runs faster than native here.

### correlation_aot.dump
```
......
     680:	62 f1 5f 08 7b df    	vcvtusi2sd %edi,%xmm4,%xmm3
     686:	45 8d 04 37          	lea    (%r15,%rsi,1),%r8d
     68a:	c5 e3 5e d9          	vdivsd %xmm1,%xmm3,%xmm3
     68e:	c5 e3 58 d8          	vaddsd %xmm0,%xmm3,%xmm3
     692:	4c 8b 8b 78 01 00 00 	mov    0x178(%rbx),%r9
     699:	c4 81 7b 11 1c 01    	vmovsd %xmm3,(%r9,%r8,1)
     69f:	45 8d 44 37 08       	lea    0x8(%r15,%rsi,1),%r8d
     6a4:	01 cf                	add    %ecx,%edi
     6a6:	62 f1 5f 08 7b df    	vcvtusi2sd %edi,%xmm4,%xmm3
......
```