wasm-micro-runtime/wasm-micro-runtime #4293: Static PGO SEGFAULTS with WAMR-2.3.0

Lukasdoe, 2025-05-23:
### Subject of the issue

When using the AOT static PGO, execution of the instrumented aot file yields: 
```
unhandled SIGSEGV, si_addr: (nil)
Aborted (core dumped)
```

### Test case

Polybench benchmarks, covariance.wasm

### Your environment

* Host OS: Ubuntu 24.04.1 LTS x86_64
* WAMR: 2.3.0, AOT, PGO

### Steps to reproduce

Compile polybench covariance benchmark with PGO enabled to AOT and execute with iwasm.

### Expected behavior

It should not crash.

### Actual behavior

It does crash.

TianlongLiang, 2025-05-30:
It's due to the WAMR upgrade of the LLVM-dependent library from 15 to 18. I am working on it and identifying the adaptation WAMR needs to accommodate the changes in LLVM for LLVM perf data. It may take some time to fix this issue, meanwhile, I tried llvm 15, it's all working fine, you can use WAMR 2.2.0 if you want to use static PGO now, if you want.

Lukasdoe, 2025-05-30:
Oh yes, I totally get why you can't just upgrade to the newest version of LLVM straight away for the whole project.

I will have a look at the parts of wamr that emit the profiling data then, thank you for the pointer! I will post any progress that I make there, maybe I can reduce the work for you as well :)

TianlongLiang, 2025-06-06:
I've got it working now; I will test it further and add a CI test for static PGO and PR soon.

closed 2025-07-10 by Lukasdoe