wasm-micro-runtime/wasm-micro-runtime #4299: Add support for PIC32M from Microchip microcontroller on FreeRTOS platform

AshishShrikhande2408, 2025-05-28:
#### Feature

Add support for PIC32M microcontroller on FreeRTOS platform

#### Benefit

Currently WAMR does not support any PIC microcontrollers from Microchip  which are widely used in automotive and Aerospace industries. Including support for PIC will enable adoption of WAMR industry wide.

#### Implementation

Since PIC32M has MIPS architecture, we can pretty much reuse code base from XTENSA, which is implemented by ESP-IDF and they also use FreeRTOS as their platform

#### Alternatives

No alternative to above.

TianlongLiang, 2025-06-03:
Hi, I don't have a MIPS environment or a PIC32M environment. Can you help identify the current gap for WAMR to support PIC32M on FreeRTOS? What's stopping WAMR from building a product-mini platform freeRTOS on a MIPS target? If just the freeRTOS platform APIs are a little out-dated, after updating them it should be fine.

AshishShrikhande2408, 2025-06-03:
I am successfully able to compile WAMR for PIC32 using freeRTOS. I need to test it tough. My configuration includes AOT enabled and I have disabled Interpreter and JIT. I think in some functions under wasm_runtime.c few conditional compilation directives are missing and I have them below:
1.  In function wasm_traverse_gc_rootset(WASMExecEnv *exec_env, void *heap)
     Error - Since interpreter is disabled we still get compiler error for "wasm_interp_traverse_gc_rootset". 
     Resolution - Added conditional compilation directive
#if WASM_ENABLE_INTERP != 0
    return wasm_interp_traverse_gc_rootset(exec_env, heap);
#endif 

2. In function wasm_call_function:
    Error -  Since interpreter is disabled we still get compiler error for "interp_call_wasm"
    Resolution - Added conditional compilation directive
#if WASM_ENABLE_INTERP != 0
    interp_call_wasm(module_inst, exec_env, function, argc, argv);
#endif

3. In function call_indirect
    Error - Since interpreter is disabled we still get compiler error for "interp_call_wasm"
    Resolution - Added conditional compilation directive
#if WASM_ENABLE_INTERP != 0
    interp_call_wasm(module_inst, exec_env, func_inst, argc, argv);
#endif

Since I am new to WAMR, Can you please confirm my approach is correct?

AshishShrikhande2408, 2025-06-04:
I am using wasm_export.h and not calling them directly. I think this issue will arise only when AOT is enabled and not interpreter. All these errors are from "wasm_runtime.c"

TianlongLiang, 2025-06-04:
It could be a CMake or build system problem, if only AOT is enabled, it shouldn't include the "wasm_runtime.c". You can see that in this cmake script file: https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/core/iwasm/interpreter/iwasm_interp.cmake

AshishShrikhande2408, 2025-06-11:
Thanks @TianlongLiang for helping me build WAMR for PIC32 on FreeRTOS. I am now facing different issue. I am able to successfully do "wasm_runtime_full_init" and  "wasm_runtime_load" but my code fails at "wasm_runtime_instantiate". The error I get is  "allocate memory failed".

TianlongLiang, 2025-06-11:
You can try increasing the size of `WASM_GLOBAL_HEAP_SIZE`, and you can also review what the initial size of the memory section inside the wasm(wast) file is, make sure the WASM_GLOBAL_HEAP_SIZE is larger than that. Or you can set verbose log level and see the log when loading the wasm, it should print out the memory section info.

AshishShrikhande2408, 2025-06-12:
ok so I was able to resolve instantiate isssue by enabling WASM_ENABLE_SHRUNK_MEMORY and by disabling WASM_MEM_ALLOC_WITH_USAGE. My memory trace is below:

**Initialize WASM runtime

Heap created, total size: 32768

   heap struct size: 356

   actual heap size: 32400

   padding bytes: 12

Register native, size: 20

Register native, size: 20

Register native, size: 20

Register native, size: 20

wasm_runtime_full_init successful!!

Run wamr with interpreter

[00:00:00:027 - 80043930]: type 0: func, param count: 1, result count: 1, ref type map count: 0

[00:00:00:035 - 80043930]: type 1: func, param count: 2, result count: 1, ref type map count: 0

[00:00:00:044 - 80043930]: type 2: func, param count: 1, result count: 0, ref type map count: 0

[00:00:00:052 - 80043930]: Load type section success.

[00:00:00:057 - 80043930]: Load import section success.

[00:00:00:063 - 80043930]: Load function section success.

[00:00:00:068 - 80043930]: Load table section success.

[00:00:00:073 - 80043930]: Load memory section success.

[00:00:00:078 - 80043930]: Load global section success.

[00:00:00:083 - 80043930]: Load export section success.

[00:00:00:088 - 80043930]: Load code segment section success.

[00:00:00:094 - 80043930]: Load data segment section success.

[00:00:00:099 - 80043930]: Found aux __data_end global, value: 1082

[00:00:00:105 - 80043930]: Found aux __heap_base global, value: 5184

[00:00:00:112 - 80043930]: Found aux stack top global, value: 5184, global index: 0, stack size: 4096

[00:00:00:121 - 80043930]: Shrink memory size to 5184

WASM module memory consumption, total size: 959

    module struct size: 280

    types size: 162

    imports size: 192

    funcs size: 73

    tables size: 40

    memories size: 16

    globals size: 96

    exports size: 48

    table segs size: 0

    data segs size: 52

    const strings size: 0

[00:00:00:152 - 80043930]: Load module success.

wasm_runtime_load successful!!

Instantiate WASM runtime

gc_heap_size = 16384

Heap created, total size: 16384

   heap struct size: 356

   actual heap size: 16016

   padding bytes: 12

[00:00:00:174 - 80043930]: Memory instantiate:

[00:00:00:178 - 80043930]:   page bytes: 13376, init pages: 1, max pages: 1

[00:00:00:184 - 80043930]:   heap offset: 5184, heap size: 8192

Heap created, total size: 8548

   heap struct size: 356

   actual heap size: 8184

   padding bytes: 8

[00:00:00:200 - 80043930]: Memory instantiate success.

WASM module inst memory consumption, total size: 0

    module inst struct size: 448

    memories size: 2634654741

        app heap size: 8192

    tables size: 4

    functions size: 160

    globals size: 96

    exports size: 8

wasm_runtime_instantiate successful!!

run main() of the application

Exec env memory consumption, total size: 9840

    exec env struct size: 16**

But it freezes while running main

AshishShrikhande2408, 2025-06-17:
I got the issue. Stack assigned to the task was less. Now I have ported classic Interpreter to PIC32 and tested with basic "Hello world" program and its working. Thanks @TianlongLiang for your help. Closing this issue. In case if I have issues will raise a new one.

closed 2025-06-17 by AshishShrikhande2408