wasm-micro-runtime/wasm-micro-runtime #4326: wasi-nn:

yamt, 2025-06-04:
we use `wasi_nn` module for wasi-nn imports by default.
i'd suggest to flip the default to use `wasi_ephemeral_nn`.
(and probably deprecate `wasi_nn` and remove it in future.)

cf. https://github.com/bytecodealliance/wasm-micro-runtime/pull/1521#discussion_r2114910503

yamt, 2025-06-17:
>  (and probably deprecate wasi_nn and remove it in future.)

wasi_ephemeral_nn support was introduced by:
```
commit db025e457afd3846815e34b385ecd79bc3262f8d
Author: liang.he <liang.he@intel.com>
Date:   Mon Jun 17 14:58:09 2024 +0800
```
and has always been recommended in our documentation.

wrt releases, it has been included since WAMR-2.1.1 (2024-07)

i wonder if anyone is still using "wasi_nn".

HongxiaWangSSSS, 2025-06-19:
> Agree.
> 
> FYI, [@HongxiaWangSSSS](https://github.com/HongxiaWangSSSS) is targeting [the same goal](https://github.com/bytecodealliance/wasm-micro-runtime/pull/4267#discussion_r2108446484). [@HongxiaWangSSSS](https://github.com/HongxiaWangSSSS) , would you mind sharing the status of the second PR?

its ok for me！
wasi_ephemeral_nn seems to have worked.

yamt, 2025-08-12:
wasi_ephemeral_nn is now the default.
build-time deprecation warning has been added too.

closed 2025-08-12 by yamt