wasm-micro-runtime/wasm-micro-runtime #4332: wasi_nn.h uses too generic names

yamt, 2025-06-05:
wasi_nn.h uses too generic names like `success` `busy`.
they are prune to confilct with unrelated things.

i'd suggest to name them as `wasi_ephemeral_nn_success` etc.
maybe it's too late to change for the legacy `wasi_nn`.
but i hope we can still change them for `wasi_ephemeral_nn`.

lum1n0us, 2025-06-08:
Those error codes are used in a wasi-nn programming context, and programmers should be able to understand what kind of error they are dealing with. But, if it is helpful and friendly, I am open to this suggestion.

yamt, 2025-06-18:
for wasi_ephemeral_nn, it was fixed by https://github.com/bytecodealliance/wasm-micro-runtime/pull/4358

for legacy wasi_nn, i guess it's too late to fix. if there are users who still want to use legacy wasi_nn, compatibility is probably more important for them.

closed 2025-06-18 by yamt