wasm-micro-runtime/wasm-micro-runtime #4363: CI: SGX 143 failure

yamt, 2025-06-13:
[spec_test_default (fast-jit, $DEFAULT_TEST_OPTIONS, ubuntu-22.04)](https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/15625115645/job/44017813763?pr=4361#logs)
```
exception is ....'type mismatch'
Writing WAST module to '/tmp/block_rauaqwbl.wast'
Compiling WASM to '/tmp/block_0ete778f.wasm'
Running: ./wabt/out/gcc/Release/wat2wasm --enable-multi-memory --enable-threads --no-check /tmp/block_rauaqwbl.wast -o /tmp/block_0ete778f.wasm
Starting interpreter for module '/tmp/block_0ete778f.wasm'
Running: ../../../product-mini/platforms/linux-sgx/enclave-sample/iwasm --heap-size=0 --repl --stack-size=131072 /tmp/block_0ete778f.wasm
module is ....'(module (func $type-break-operand-f64-vs-f32 (result f64)
Error: Process completed with exit code 143.
```

references:
https://github.com/actions/runner-images/issues/6680
https://github.com/bytecodealliance/wasm-micro-runtime/pull/4222 this doesn't seem working well?

lum1n0us, 2025-06-17:
Yes, it seems the retry approach based on bash can't detect the exit; maybe using the gh command is better.

closed 2025-06-20 by lum1n0us

lum1n0us, 2025-06-26:
retry-action doesn't work. https://github.com/bytecodealliance/wasm-micro-runtime/actions/runs/15891351369/job/44814369031?pr=4415

what's our next option?

yamt, 2025-06-27:
> what's our next option?

i suspect the 143 failure is something which can't be detected by the workflow step itself.

i wonder why it only happens on the SGX workflow frequently.
if it's because of the size, maybe we can try to reduce/divide the workflow.

lum1n0us, 2025-06-27:
> i wonder why it only happens on the SGX workflow frequently.

No idea. I used to think it might be a compatibility problem, but found nothing after going through the manuals.

> if it's because of the size, maybe we can try to reduce/divide the workflow.

Are you suggesting that SGX installation might consume too much resource and not leave enough disk space for CI? 

Regarding reduce/divide, I thought jobs, especially spec-tests, are parallel and divided across different machines.

yamt, 2025-06-27:
> > if it's because of the size, maybe we can try to reduce/divide the workflow.
> 
> Are you suggesting that SGX installation might consume too much resource and not leave enough disk space for CI?

my assumption was the job was somehow killed because of some resource limit.
i have no idea if it's the case or what resource it is. it can be disk, cpu, memory, swap, etc.