wasm-micro-runtime/wasm-micro-runtime #4364: wasi configurations in wasm_module_t

yamt, 2025-06-13:
`wasm_runtime_set_wasi_args` / `wasm_runtime_set_wasi_args_ex` sets wasi configurations on `wasm_module_t`.
it isn't intuitive because those configuartions logically belong to `wasm_module_inst_t`.
ideally it would be nice to allow to instantiate multiple `wasm_module_inst_t` from a single `wasm_module_t`.

one way to fix it is to make them a part of `InstantiationArgs`.
it's unfortunate `InstantiationArgs` is defined in a way which can't be extended without breaking the ABI.

(i thought i filed this issue long time ago. but i couldn't find it. sorry if it's duplicated.)

lum1n0us, 2025-06-18:
You are right. The current design doesn't accommodate usage scenarios like varying WASI contexts for multiple instances (either continuously in one thread or parallelly across multiple threads) of a module.

I prefer to add a new API like `wasm_runtime_instantiate_with_wasi(wasm_module_t, InstantiationArgs*, WasiConfigArgs*, char*, uint32_t)` and a new set of helpers for building `WasiConfigArgs`.

lum1n0us, 2025-06-23:
#4256

lum1n0us, 2025-07-02:
Looks good to me. Let's see how it goes.