wasm-micro-runtime/wasm-micro-runtime #4378: opendir failed on Zephyr platform

banana-sun, 2025-06-17:
OS: Zephyr v3.7.1
WAMR brance: https://github.com/bytecodealliance/wasm-micro-runtime/tree/dev/zephyr_file_socket
Hardware: NUCLEO-H563ZI
Zephyr File System: lfs/test.txt
WAMR FS interaction folder: --dir=lfs/

Application code:

```c
#include <stdio.h>
#include <stdlib.h>

#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <unistd.h>
#include <string.h>
#include <errno.h>
#include <dirent.h>

void list_dir(const char *path)
{
    struct dirent *entry;
    DIR *dp;

    dp = opendir(path);
    if (dp == NULL) {
        printf("opendir %s\r\n", strerror(errno));
        return;
    }

    while ((entry = readdir(dp))) {
        printf("%s ", entry->d_name);
        if (DT_DIR == entry->d_type) {
            printf("dir\n");
        }
        else if (DT_REG == entry->d_type) {
            printf("file\n");
        }
        else {
            printf("other\n");
        }
    }

    closedir(dp);
}

int
main(int argc, char **argv)
{
    printf("Hello, I'm WASM application!\n");

    list_dir("/lfs");
    return 0;
}
```

banana-sun, 2025-06-22:
I tried again using the latest version of this PR(https://github.com/bytecodealliance/wasm-micro-runtime/pull/4377), but still had this problem.

lucasAbadFr, 2025-06-30:
Clearly, the behavior is implemented this way by design. As for whether it should be considered a problem. I don’t think I’m the right person to make that call.

When I worked on the file system integration, I focused on maintaining a 1:1 behavior (to a resonable limit) between my Zephyr samples and the Linux `iwasm` implementation, which is currently the most complete and mature target. The goal was to ensure the same WASM module would behave consistently across both platforms.

For more context on the design decisions, including whether a preopened directory should be directly `open()`'able, or whether this behavior could (or should) evolve. I think @wenyongh or @luminous might be able to offer better insights than me.

Hope this helps !