wasm-micro-runtime/wasm-micro-runtime #4386: os_mutex_trylock

yamt, 2025-06-18:
our platform api doesn't have os_mutex_trylock.
is it intentional? eg. impossible/inefficient for some platforms?

TianlongLiang, 2025-06-18:
I think it's intentional, but not for the reason that it's impossible/inefficient for some platforms. The other APIs, such as `os_mutex_lock` in `platform_api_vmcore.h` are required platform APIs to support a certain WAMR feature, in this case, the thread manager, and we don't need os_mutex_trylock to do that; it's not a must, so we don't add it.

yamt, 2025-06-30:
> I think it's intentional, but not for the reason that it's impossible/inefficient for some platforms. The other APIs, such as `os_mutex_lock` in `platform_api_vmcore.h` are required platform APIs to [...]

ok.
it makes sense to keep the platform api minimum.

for the record, i recently happened to have a use case. (https://github.com/bytecodealliance/wasm-micro-runtime/pull/4387)
i don't claim this use case alone is enough reason to add trylock though.

maybe we can revisit when/if more use cases come up.
i'm going to close this for now.

closed 2025-06-30 by yamt