wasm-micro-runtime/wasm-micro-runtime #4443: error reporting of wasm_export.h apis are not machine-friendly

yamt, 2025-07-02:
many of apis in wasm_export.h report errors via `char *error_buf, uint32_t error_buf_size`.
while it's convenient when you just want to report the error to human beings and bail out,
it isn't ideal for embedders which wants to handle errors in more sophisticated ways.

for example, consider an embedder which wants to know if the failure was retry-able or not.
with the current api, it has to parse the error string to make a wild guess what was the cause.

lum1n0us, 2025-07-03:
I actually feel the same way, but it seems there are very few options for us to improve that without disrupting current APIs.
- Implement a Linux-like error number system, allowing us to retrieve an error number when APIs return false.
- Standardize error messages by defining a set of error messages and enabling retrieval of error numbers from these messages.
- Expand the return type from `bool` to an integer for kind of errno (or a pointer of a structure contains error information)

yamt, 2025-07-04:
> I actually feel the same way, but it seems there are very few options for us to improve that without disrupting current APIs.
> 
>     * Implement a Linux-like error number system, allowing us to retrieve an error number when APIs return false.

i'm afraid we have to store the last error in a global variable, at least for some of the current apis.

>     * Standardize error messages by defining a set of error messages and enabling retrieval of error numbers from these messages.

i guess this is the most conservative option.

maybe we can start using messages like "E0123 foo bar".

and
```
int wasm_runtime_extract_error_code(const char *errormsg)
```
which returns 123 for the above example.

> 
>     * Expand the return type from `bool` to an integer for kind of errno (or a pointer of a structure contains error information)

it sounds like the most reasonable option in case we decide to break the compatibility.