wasm-micro-runtime/wasm-micro-runtime #4454: [Performance] Abnormal execution time in Fast JIT mode

gaaraw, 2025-07-06:
### Subject of the issue

Hello, when I tried to execute some benchmark programs using WAMR's three JIT modes, I experienced performance degradation and abnormal runtime in `Fast JIT` mode. To do this, I also used two other JIT runtime tools to execute the same program to confirm. The specific performance is as follows:

[image]

The data in the graph is measured in seconds, and each data is the result of ten executions and averaged.

### Test case

[test_case.zip](https://github.com/user-attachments/files/21087210/test_case.zip)

### Your environment

* WAMR version：iwasm 2.3.1
* wasmer version：wasmer 6.0.1
* wasmtime version：wasmtime 35.0.0 (9c2e6f17c 2025-06-17)
* emcc version：4.0.10 (b7dc6e5747465580df5984e723b9d1f10d8e804b)
* Host OS：Ubuntu 22.04.5 LTS 64位
* CPU：12th Gen Intel® Core™ i7-12700 × 20

Both `wasmer` and `wasmtime` are release build by default, and the `wasmer` backend is the default Cranelift.

### Steps to reproduce

```
# Compile C file (the same goes for other files)
emcc flops-1.c -o flops-1.wasm -O2 -s WASM=1
# Execute the wasm file and collect the data
perf stat -r 10 -e 'task-clock' /path/to/build_fast_jit/iwasm flops-1.wasm
perf stat -r 10 -e 'task-clock' /path/to/build_llvm_jit/iwasm flops-1.wasm
perf stat -r 10 -e 'task-clock' /path/to/build_mul_jit/iwasm flops-1.wasm
perf stat -r 10 -e 'task-clock' /path/to/wasmer run flops-1.wasm
perf stat -r 10 -e 'task-clock' /path/to/wasmtime flops-1.wasm
```

### Expected and Actual behavior

`Fast JIT` should have a better execution time than `llvm JIT` and `multi-tier JIT`, at least within 1.5 times. and not up to 7 times (e.g., flops-1.wasm, flops-4.wasm, flops-5.wasm). Several other programs, although relatively less anomalous, are also about 3-4 times as high.

### Extra Info

These benchmarks are for floating-point performance, I wonder if there's an issue in `Fast JIT`? Also regarding the other two runtime tools, `wasmer` and `wasmtime`, both seem to perform better than WAMR's three JIT modes, is this due to different backends, or is it because I ignored some parameter settings? If you can please answer, thank you!
If you need any other relevant information, please let me know and I will do my best to provide it. Looking forward to hearing from you! Thank you!

TianlongLiang, 2025-07-07:
> `Fast JIT` should have a better execution time than `llvm JIT` and `multi-tier JIT`

Not really, it's called "fast" in a way that it compiles to jitted code faster, not executes faster, the llvm jit definitely has a faster execution time. It's expected that it's slower, but maybe not by that much, I will find some past data to confirm what's normal like.

And given your measuring commands:
```bash
perf stat -r 10 -e 'task-clock' /path/to/build_fast_jit/iwasm flops-1.wasm
```
I think you are not only measuring execution time(unless you only count the jitted function data in your table), but also the time loading wasm file and the time generating jitted code. I don't think we can simply draw the conclusion that this data represents performance.

gaaraw, 2025-07-08:
First of all, thank you for your reply. I'm also trying to confirm the normal relative execution time.

Regarding the `perf stat` command, what I collect is indeed the time of the overall process execution, and I am now looking for ways to collect the time spent on the different stages in detail (if you have any suggestions I would be grateful).

Also, if you are considerate, the results of the `perf stat` command are indeed not simply representative of performance. However, I think the results are useful for considering relative performance. The reason why the time-consuming nature of these procedures caught my attention was the sudden increase in their relative time-consuming.

syrusakbary, 2025-07-09:
I'm surprised with the wasmer times here. We should be 2-3x faster than Wasmtime (when using the LLVM backend).

Are these benchmarks using wasmer+llvm or wasmer+cranelift?

gaaraw, 2025-07-10:
> Are these benchmarks using wasmer+llvm or wasmer+cranelift?

wasmer+cranelift.

But, I just used the command:
```
perf stat -r 10 -e 'task-clock' /path/to/wasmer run flops-1.wasm --llvm
```
Here is the result (it looks similar to cranelift):

[image]