wasm-micro-runtime/wasm-micro-runtime #4480: WAMR JIT mode incorrectly outputs type_mismatch

teyahb8, 2025-07-10:
### Subject of the issue

I executed the following test case with different runtimes. Other runtimes supporting typed function references would output the same number, including WAMR's AOT mode. But, WAMR's JIT mode outputs "type mismatch" error. Other runtime modes that don't support typed function references would output that "typed function references are not supported."

### Test case

```
(module
  (type (;0;) (func))
  (type (;1;) (func (result v128)))
  (export "main" (func 0))
  (export "to_test" (func 0))
  (func (;0;) (type 1) (result v128)
    v128.const i32x4 0x00000000 0x00000000 0x00000000 0xfd000000
    br 0
    unreachable
    unreachable
    unreachable
    unreachable
    unreachable
    unreachable
    unreachable
    unreachable
    unreachable
    unreachable
    i64.or
    ref.null 0
    unreachable
    i16x8.lt_u
  )
)
```

### Your environment

* Host OS: Ubuntu 22.04
* WAMR version (2.3.1), platform (Linux), cpu architecture (x86_64), running mode (JIT).

### Steps to reproduce

Run the command: `iwasm --llvm-jit -f main i16x8.lt_u_15_10_1752003706322386.wasm`

### Expected behavior

Should output "336294682933583715844663186250927177728".

### Actual behavior

Output is "type mismatch"

### Extra Info

Can you please confirm if this is a bug in WAMR's JIT mode? Thanks in advance.

Septa2112, 2025-07-25:
I seem to have encountered some issues while trying to reproduce this. If possible, could you attach the `wasm` binary you tested and share some more detailed steps, such as the compilation options you used? That would be helpful.

teyahb8, 2025-07-25:
Thanks for the feedback. Here are the compilation options:

```
cmake -DWAMR_BUILD_BULK_MEMORY=1 -DWAMR_BUILD_REF_TYPES=1 -DWAMR_BUILD_SIMD=1 -DWAMR_BUILD_JIT=1 -DLLVM_DIR=/usr/lib/llvm-18/lib/cmake/llvm ..
make -j
```

And, here is the testcase in .zip format since .wasm file could not be uploaded here.

**Testcase:** [testcase.zip](https://github.com/user-attachments/files/21432858/testcase.zip)

teyahb8, 2025-07-30:
Thanks for confirming with the nice explanation.