wasm-micro-runtime/wasm-micro-runtime #4482: Likely validation bug related to memory alignment.

teyahb8, 2025-07-11:
### Subject of the issue

I executed the following test case on different runtimes. Other runtimes that support the "Exception Handling" proposal would output "Exception: unreachable". But, after building WAMR with "Exception Handling" proposal, both WAMR AOT and JIT mode outputs a validation error "alignment must not be larger than natural".

In the test case, there is an instruction `i32.store offset=1916 align=2`. If the validation error occurs due to the alignment value on this instruction, can you please recheck if there indeed should be a validation error? According to the specification of store operation, "The alignment 2^memarg.align must not be larger than the bit width of 𝑡 divided by 8." For an i32 instruction, 𝑡 divided by 8 should be 4, and 2^(align=2) should also be 4. So, this seems to be a valid alignment.

### Test case

```
(module
  (type (;0;) (func (param i32 i32 i32 i32) (result i32)))
  (type (;1;) (func (param i32)))
  (type (;2;) (func))
  (type (;3;) (func (result i32)))
  (table (;0;) 10 20 funcref)
  (memory (;0;) 1 5)
  (global (;0;) i32 i32.const 541)
  (global (;1;) (mut i32) i32.const 191)
  (global (;2;) f32 f32.const 0x1.0e8p+9 (;=541;))
  (global (;3;) (mut f32) f32.const 0x1.8p+7 (;=192;))
  (global (;4;) i64 i64.const 54)
  (global (;5;) (mut i64) i64.const 19)
  (global (;6;) f64 f64.const 0x1.bp+5 (;=54;))
  (global (;7;) (mut f64) f64.const 0x1.3p+4 (;=19;))
  (global (;8;) (mut i32) i32.const 0)
  (global (;9;) (mut f32) f32.const 0x0p+0 (;=0;))
  (global (;10;) (mut i64) i64.const 0)
  (global (;11;) (mut f64) f64.const 0x0p+0 (;=0;))
  (export "main" (func 0))
  (export "to_test" (func 0))
  (elem (;0;) (i32.const 0) func 1 2 3 0 1)
  (elem (;1;) (i32.const 5) func 3 3 2 1)
  (elem (;2;) (i32.const 9) func 0)
  (func (;0;) (type 2)
    (local i32 f32 i64 f64)
    i32.const 305419896
    local.set 0
    f32.const 0x1.8cp+6 (;=99;)
    local.set 1
    i64.const -72057589709208571
    local.set 2
    f64.const 0x1.5f0b08c960a79p+109 (;=890000000000000000000000000000000;)
    local.set 3
    unreachable
    i32.const -62
    unreachable
    i32.store offset=1916 align=2
    unreachable
    throw_ref
    local.get 0
    global.set 8
    local.get 1
    global.set 9
    local.get 2
    global.set 10
    local.get 3
    global.set 11
  )
  (func (;1;) (type 3) (result i32)
    i32.const 1
    i32.const 3
    i32.add
  )
  (func (;2;) (type 3) (result i32)
    i32.const 2
    i32.const 3
    i32.add
  )
  (func (;3;) (type 3) (result i32)
    i32.const 3
    i32.const 3
    i32.add
  )
  (data (;0;) (i32.const 8) "\01\02\03\04\05\06\07\08")
  (data (;1;) (i32.const 16) "\01\02\03\04\05\06\07\08\ff")
  (data (;2;) (i32.const 32) "\01\02\03\04\05\06\07\08\ff")
)
```

### Your environment

* Host OS (Ubuntu 22.04)
* WAMR version (2.3.1), platform (Linux), cpu architecture (x86_64), running mode (AOT/JIT).

### Steps to reproduce

1. AOT: `wamrc --bounds-checks=1 -o test.aot test.wasm`; `iwasm --heap-size=0 -f to_test test.aot`
2. JIT: `iwasm --heap-size=0 --llvm-jit -f to_test test.wasm` 

### Expected behavior

Output should be: "Exception: unreachable"

### Actual behavior

Actually outputs: "WASM module load failed: invalid memop flags: alignment must not be larger than natural"

### Extra Info

Can you please confirm if this looks like a validation bug? Thanks in advance.

teyahb8, 2025-07-29:
I have used `wasm-tools`. Specifically, I have used the following command: `wasm-tools parse -o test.wasm test.wat`.

teyahb8, 2025-07-29:
I am also attaching the WASM binary here. I renamed the file to `test.txt` since `.wasm` files are not allowed to be uploaded here. 

[test.txt](https://github.com/user-attachments/files/21491792/test.txt)

Septa2112, 2025-07-31:
When I run the original wasm binary you uploaded, I will get the same error result as you. However, when I convert your `.wat` file into a wasm binary using `wasm-tools` and run it, the alignment-related issue does not occur.

I tried dumping both wasm binaries for comparison. I found that

### Dump Result

#### test.wasm (the original WASM binary you uploaded)
```
 0000f8: 36 41 00 fc 0e             | i32.store 1 1916
```
The original wasm binary looks like this. In WAMR, the `align` value is not read using LEB128 encoding. So the value of  `align` will be read as `0x41` in this case. This exceeds the boundary condition and triggers the error. I don't think this is an issue with WAMR.

#### test.wasm -> test.wat -> test_new.wasm (converted by `wasm-tools`)
```
 0000f8: 36 01 fc 0e                | i32.store 1 1916
```
When the original wasm binary is converted to `.wat` file by `wasm-tools`, the bytes `0x41 0x00` are interpreted as LEB128 which is decimal 1. That's why the `.wat` file shows `align=2`. Then I convert the `.wat` file to `test_new.wasm`, the dump result of new wasm binary will show `36 01 fc 0e`.

teyahb8, 2025-07-31:
Thanks for the explanation. Originally, I had generated the .wasm binary test case. I later converted it to .wat format for reporting. The wasm -> wat -> wasm conversion might be inadvertently removing the error causing part from the test case. But, when I run the original wasm without the wat/wasm conversions, other runtimes would output `Exception: unreachable` and wamr runtime would incorrectly report the `alignment error`. I think regardless of a back-to-back conversion between wat and wasm, the output of the wamr runtime should be consistent with the other runtimes for the original test case.

lum1n0us, 2025-07-31:
Just out of curiosity, may I ask how you obtained the original .wasm? Was it compiled from a guest language, compiled from wat, or modified from another .wasm binary?