wasm-micro-runtime/wasm-micro-runtime #4501: Missing Subtyping Depth Validation

Q1IQ, 2025-07-17:
### Subject of the issue

iwasm does not enforce the subtyping depth limit when compiling WebAssembly modules, which leads to successful execution of modules that should otherwise fail validation due to excessive subtyping depth. This behavior is inconsistent with other engines like V8 and WebKit, which enforce a subtyping depth limit of 64.

### Test case

[1.wasm.txt](https://github.com/user-attachments/files/21302526/1.wasm.txt)

### Your environment

* OS: Ubuntu 20.04
* CPU: amd64
* WAMR version: iwasm 2.2.0
* Commands:
```
iwasm --interp -f main ./program.wasm
```

### Actual behavior
```
0x1:i32
```

### Expected State

The module should fail validation due to the subtyping depth exceeding 64, aligning with the behavior observed in V8 and WebKit. An appropriate error message should indicate that the type depth exceeds the allowed limit, similar to:
```
subtype depth for Type section's 64th signature exceeded the limits of 63
```

TianlongLiang, 2025-07-21:
If I remember correctly, the gc proposal itself didn't limit the subtyping depth, but I do think it's a good idea to set a limitation in the implementation to prevent potential overflow.

Q1IQ, 2025-07-23:
Thanks for the comment! For reference, the current spec on this can be found in the GC proposal [here](https://github.com/WebAssembly/gc/blob/main/document/js-api/index.bs#L1443).

TianlongLiang, 2025-07-23:
It's not in the core spec; the link you refer to is related to js-api and is listed as "Implementation-defined Limits." The language specification itself https://webassembly.github.io/gc/core/intro/overview.html didn't set a specific limitation