wasm-micro-runtime/wasm-micro-runtime #4525: WAMR AOT incorrectly outputs `out of bounds` exception

teyahb8, 2025-07-30:
### Subject of the issue

I generated a test case and executed it with different runtimes. Other runtimes like wasmtime, wasmer, wasmedge, and wamr's JIT mode all output a list of numbers. But, wamr's AOT mode outputs `out of bounds` exception.

### Test case

```
(module
  (type (;0;) (func (result v128 v128 v128 v128 v128 v128 v128)))
  (memory (;0;) 65536 65536)
  (global (;0;) (mut v128) v128.const i32x4 0x00000000 0x00000000 0x00000000 0x00000000)
  (export "main" (func 0))
  (func (;0;) (type 0) (result v128 v128 v128 v128 v128 v128 v128)
    (local v128)
    local.get 0
    global.get 0
    local.get 0
    local.get 0
    v128.const i32x4 0xeed158ea 0xca9d03b8 0xd7e356de 0xb508ae66
    loop ;; label = @1
      v128.const i32x4 0x00000000 0x00000000 0x00000000 0x00000000
      local.set 0
      local.get 0
      local.get 0
      f32x4.div
      local.set 0
      local.get 0
      local.get 0
      local.get 0
      local.get 0
      i32.const 6506
      i32x4.shr_s
      i32x4.min_u
      i8x16.extract_lane_s 1
      v128.load32x2_u offset=6198 align=1
      i32x4.extmul_low_i16x8_u
      i64x2.sub
      local.get 0
      i32x4.eq
      local.tee 0
      local.get 0
      i32x4.lt_u
      local.tee 0
      v128.const i32x4 0x38bc50fa 0x761038d3 0xfd7b04bf 0x239cac85
      i8x16.ge_s
      local.get 0
      i32x4.max_u
      local.tee 0
      local.get 0
      i64x2.ne
      local.tee 0
      local.get 0
      f64x2.pmax
      local.tee 0
      v128.const i32x4 0x8e848664 0xb0a665ed 0x5486b3e2 0x6ab0d25d
      i8x16.gt_s
      local.get 0
      i8x16.narrow_i16x8_u
      local.set 0
      i32.const 0
      br_if 0 (;@1;)
    end
    v128.const i32x4 0xf1905511 0xfa983748 0x9a69bb75 0xd7b7c5f7
    local.get 0
    local.get 0
    local.get 0
    local.get 0
    local.get 0
    v128.const i32x4 0x00000000 0x00000000 0x00000000 0x00000000
    local.get 0
    local.get 0
    return
  )
)
```

### Your environment

* Host OS (Ubuntu 22.04)
* WAMR version (2.3.1), platform (Linux), cpu architecture (x86_64), running mode (AOT), etc.

### Steps to reproduce

1. Convert wat test case to wasm file
2. Compile wasm file to AOT file: `wamrc --bounds-checks=1 -o shrunken_test306117.aot shrunken_test306117.wasm`
3. Run: `iwasm --heap-size=0 -f main shrunken_test306117.aot`

### Expected behavior

```
<0x0000000000000000 0x0000000000000000>:v128,<0x0000000000000000 0x0000000000000000>:v128,<0x0000000000000000 0x0000000000000000>:v128,<0x0000000000000000 0x0000000000000000>:v128,<0x0000000000000000 0x0000000000000000>:v128,<0x0000000000000000 0x0000000000000000>:v128,<0x0000000000000000 0x0000000000000000>:v128
```

### Actual behavior

```
Exception: out of bounds memory access
```

### Extra Info

Can you please confirm? Thanks in advance.

TianlongLiang, 2025-07-30:
This has something to do with LLVM optimization, `--opt-level=0` would make the behavior consistent. I will look into it further and keep you updated

teyahb8, 2025-07-30:
Thanks for confirming, @TianlongLiang.