wasm-micro-runtime/wasm-micro-runtime #4539: WAMR AOT mode crashes

teyahb8, 2025-08-05:
### Subject of the issue

I have generated a test case that I executed it with wamr runtime. The AOT mode of wamr runtime crashes. This crash occurs when `wamrc` is built with `BUILD_WAMR_LIBC_WASI=1` and `iwasm` is built with `BUILD_WAMR_LIBC_WASI=0`.

### Test case

```
(module
  (type (;0;) (func))
  (type (;1;) (func (param i32 i32) (result i32)))
  (type (;2;) (func (param i32)))
  (import "wasi_snapshot_preview1" "fd_renumber" (func (;0;) (type 1)))
  (import "wasi_snapshot_preview1" "proc_exit" (func (;1;) (type 2)))
  (memory (;0;) 2)
  (export "memory" (memory 0))
  (export "_start" (func 2))
  (func (;2;) (type 0)
    i32.const 3
    i32.const 3
    call 0
    call 1
  )
)
```

### Your environment

* Host OS (Ubuntu 22.04)
* WAMR version (2.3.1), platform (Linux), cpu architecture (x86_64), running mode (AOT), etc.

### Steps to reproduce

1. Convert wat to wasm.
2. Run: `wamrc --bounds-checks=1 -o test.aot test.wasm`
3. Run: `iwasm --heap-size=0 -f _start test.aot`

### Expected behavior

Should exit without crashing. If the user is expected to build both `wamrc` and `iwasm` with the same value for the `BUILD_WAMR_LIBC_WASI` flag, an error message should be helpful.

### Actual behavior

```
[12:16:33:373 - 7FFFF7F337C0]: warning: failed to link import function (wasi_snapshot_preview1, fd_renumber)
[12:16:33:373 - 7FFFF7F337C0]: warning: failed to link import function (wasi_snapshot_preview1, proc_exit)
unhandled SIGSEGV, si_addr: (nil)
Aborted
```

### Extra Info

Here is the stack trace from the crash:

```
* thread #1, name = 'iwasm', stop reason = signal SIGSEGV: invalid address (fault address: 0x0)
  * frame #0: 0x0000000000000000
    frame #1: 0x0000000040002051
    frame #2: 0x00007ffff5610545 libiwasm.so.2`invoke_native_with_hw_bound_check(exec_env=0x00005555555a03f0, func_ptr=0x0000000040002004, func_type=0x000055555559ffa0, signature=0x0000000000000000, attachment=0x0000000000000000, argv=0x00005555555a00b0, argc=0, argv_ret=0x00005555555a00b0) at aot_runtime.c:2471:13
    frame #3: 0x00007ffff5610cea libiwasm.so.2`aot_call_function(exec_env=0x00005555555a03f0, function=0x000055555559ff30, argc=0, argv=0x00005555555a00b0) at aot_runtime.c:2748:15
    frame #4: 0x00007ffff55cf113 libiwasm.so.2`wasm_runtime_call_wasm(exec_env=0x00005555555a03f0, function=0x000055555559ff30, argc=0, argv=0x00005555555a00b0) at wasm_runtime_common.c:2671:15
    frame #5: 0x00007ffff55ba309 libiwasm.so.2`execute_func(module_inst=0x00005555555a0160, name="_start", argc=0, argv=0x00007fffffffddc0) at wasm_application.c:685:10
    frame #6: 0x00007ffff55ba782 libiwasm.so.2`wasm_application_execute_func(module_inst=0x00005555555a0160, name="_start", argc=0, argv=0x00007fffffffddc0) at wasm_application.c:914:11
    frame #7: 0x0000555555556bf2 iwasm`app_instance_func(module_inst=0x00005555555a0160, func_name="_start") at main.c:137:5
    frame #8: 0x0000555555558352 iwasm`main(argc=1, argv=0x00007fffffffddb8) at main.c:1050:21
    frame #9: 0x00007ffff4a29d90 libc.so.6`__libc_start_call_main(main=(iwasm`main at main.c:573:1), argc=5, argv=0x00007fffffffdd98) at libc_start_call_main.h:58:16
    frame #10: 0x00007ffff4a29e40 libc.so.6`__libc_start_main_impl(main=<unavailable>, argc=5, argv=<unavailable>, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd88) at libc-start.c:392:3
    frame #11: 0x0000555555556725 iwasm`_start + 37
```

Can you please confirm? Thanks in advance.

Septa2112, 2025-08-06:
I'm investigating the root cause and attempting to fix this crash. I'll update here if there's any progress.

teyahb8, 2025-08-06:
Thanks for the confirmation.