wasm-micro-runtime/wasm-micro-runtime #4544: WAMR AOT and JIT modes hang indefinitely

teyahb8, 2025-08-05:
### Subject of the issue

I have generated a test case and executed it with different runtimes. Other runtimes like wasmtime, wasmer's cranelift backend, and wasmedge all output `out of bounds` exception. But, wamr's both AOT and JIT modes hang indefinitely. It might be worth noting issue #4481 where only the JIT mode wouldn't show `out of bounds` exception likely due to an optimization bug. But, with this test case, in addition to JIT mode, AOT mode also wouldn't show `out of bounds` exception. 

### Test case

[wasm_testcase_file](https://github.com/user-attachments/files/21603395/shrunken_test128084.txt)

```
(module
  (type (;0;) (func (result v128)))
  (type (;1;) (func (result v128)))
  (memory (;0;) 65536 65536)
  (export "main" (func 1))
  (func (;0;) (type 0) (result v128)
    unreachable
  )
  (func (;1;) (type 1) (result v128)
    (local v128)
    i32.const 0
    local.get 0
    v128.store64_lane offset=2437 align=1 1
    i32.const 0
    local.get 0
    v128.store offset=9865 align=1
    i32.const 1
    local.get 0
    v128.store16_lane offset=3276 align=1 1
    i32.const 0
    local.get 0
    v128.store offset=2908 align=1
    block ;; label = @1
      loop ;; label = @2
        i32.const 0
        br_if 1 (;@1;)
        i32.const 0
        local.get 0
        v128.store offset=1155 align=1
        block ;; label = @3
          block ;; label = @4
            i32.const 1
            br_if 0 (;@4;)
            br 1 (;@3;)
          end
          v128.const i32x4 0x00000000 0x00000000 0x00000000 0x00000000
          local.set 0
          local.get 0
          local.get 0
          f32x4.div
          local.set 0
          local.get 0
          i16x8.extract_lane_s 1
          v128.load8x8_u offset=9759 align=1
          local.set 0
        end
        i32.const 0
        i8x16.splat
        local.set 0
        i32.const 0
        local.get 0
        v128.store16_lane offset=3254 align=1 0
        i32.const 0
        local.get 0
        v128.store64_lane offset=4332 align=1 1
        br 0 (;@2;)
      end
    end
    call 0
    i32.const 0
    local.get 0
    v128.store32_lane offset=9308 align=1 0
    local.get 0
    return
    local.get 0
    return
  )
)
```

### Your environment

* Host OS (Ubuntu 22.04)
* WAMR version (2.3.1), platform (Linux), cpu architecture (x86_64), running mode (AOT/JIT), etc.

### Steps to reproduce

1. Convert wat to wasm or use the attached wasm file.
2. Run: `wamrc --bounds-checks=1 -o test.aot test.wasm`
3. Run: `iwasm --heap-size=0 -f main test.aot`
4. Run: `iwasm --heap-size=0 --llvm-jit -f main test.wasm`

### Expected behavior

Should output: `out of bounds` exception.

### Actual behavior

Hangs indefinitely.

### Extra Info

Can you please confirm? Thanks in advance.

TianlongLiang, 2025-08-06:
This is also because of LLVM optimization, and the reason this issue occur in both LLVM jit and aot is that mostly like a shared optimization pass cause it.
If you run with 
```bash
$WAMRC --bounds-checks=1 --opt-level=0 -o test.aot test.wasm
``` 
Then it's oob

If you output the llvm ir with opt:
```llvmir
; ModuleID = 'WASM Module'
source_filename = "WASM Module"
target datalayout = "e-m:e-p270:32:32-p271:32:32-p272:64:64-i64:64-i128:128-f80:128-n8:16:32:64-S128"
target triple = "x86_64-unknown-linux-gnu"

@aot_stack_sizes = internal global [2 x i32] [i32 -1, i32 -1], section ".aot_stack_sizes"

@aot_stack_sizes_alias = alias [2 x i32], ptr @aot_stack_sizes

; Function Attrs: noinline
define noundef <2 x i64> @"aot_func#0"(ptr nocapture readonly %exec_env) local_unnamed_addr #0 {
stack_bound_check_block:
  %stack_bound_addr = getelementptr inbounds ptr, ptr %exec_env, i64 4
  %native_stack_bound = load ptr, ptr %stack_bound_addr, align 8
  %sp_ptr = alloca i32, align 4
  %sp = ptrtoint ptr %sp_ptr to i64
  %size32 = load i32, ptr @aot_stack_sizes, align 4
  %size = zext i32 %size32 to i64
  %underflow = icmp ult i64 %sp, %size
  %new_sp = sub i64 %sp, %size
  %bound_base_int = ptrtoint ptr %native_stack_bound to i64
  %cmp = icmp ult i64 %new_sp, %bound_base_int
  %cmp2 = or i1 %underflow, %cmp
  br i1 %cmp2, label %got_exception, label %call_wrapped_func

call_wrapped_func:                                ; preds = %stack_bound_check_block
  %tail_call = musttail call <2 x i64> @"aot_func_internal#0"(ptr %exec_env)
  ret <2 x i64> %tail_call

got_exception:                                    ; preds = %stack_bound_check_block
  %aot_inst_addr = getelementptr inbounds ptr, ptr %exec_env, i64 2
  %aot_inst = load ptr, ptr %aot_inst_addr, align 8
  call void @aot_set_exception_with_id(ptr %aot_inst, i32 11)
  ret <2 x i64> zeroinitializer
}

; Function Attrs: noinline
define internal noundef <2 x i64> @"aot_func_internal#0"(ptr nocapture readonly %exec_env) unnamed_addr #0 {
got_exception:
  %aot_inst_addr = getelementptr inbounds ptr, ptr %exec_env, i64 2
  %aot_inst = load ptr, ptr %aot_inst_addr, align 8
  tail call void @aot_set_exception_with_id(ptr %aot_inst, i32 0)
  ret <2 x i64> zeroinitializer
}

declare void @aot_set_exception_with_id(ptr, i32) local_unnamed_addr

; Function Attrs: noinline
define noundef <2 x i64> @"aot_func#1"(ptr nocapture readonly %exec_env) local_unnamed_addr #0 {
stack_bound_check_block:
  %stack_bound_addr = getelementptr inbounds ptr, ptr %exec_env, i64 4
  %native_stack_bound = load ptr, ptr %stack_bound_addr, align 8
  %sp_ptr = alloca i32, align 4
  %sp = ptrtoint ptr %sp_ptr to i64
  %size32 = load i32, ptr getelementptr inbounds ([2 x i32], ptr @aot_stack_sizes, i64 0, i64 1), align 4
  %size = zext i32 %size32 to i64
  %underflow = icmp ult i64 %sp, %size
  %new_sp = sub i64 %sp, %size
  %bound_base_int = ptrtoint ptr %native_stack_bound to i64
  %cmp = icmp ult i64 %new_sp, %bound_base_int
  %cmp2 = or i1 %underflow, %cmp
  br i1 %cmp2, label %got_exception, label %call_wrapped_func

call_wrapped_func:                                ; preds = %stack_bound_check_block
  %tail_call = musttail call <2 x i64> @"aot_func_internal#1"(ptr %exec_env)
  ret <2 x i64> %tail_call

got_exception:                                    ; preds = %stack_bound_check_block
  %aot_inst_addr = getelementptr inbounds ptr, ptr %exec_env, i64 2
  %aot_inst = load ptr, ptr %aot_inst_addr, align 8
  call void @aot_set_exception_with_id(ptr %aot_inst, i32 11)
  ret <2 x i64> zeroinitializer
}

; Function Attrs: noinline
define internal noundef <2 x i64> @"aot_func_internal#1"(ptr nocapture readonly %exec_env) unnamed_addr #0 {
block0_begin:
  %aot_inst_addr = getelementptr inbounds ptr, ptr %exec_env, i64 2
  %aot_inst23 = load i64, ptr %aot_inst_addr, align 8
  %0 = inttoptr i64 %aot_inst23 to ptr
  %1 = getelementptr inbounds i8, ptr %0, i64 376
  %mem_base_addr26 = load i64, ptr %1, align 8
  %2 = inttoptr i64 %mem_base_addr26 to ptr
  %3 = getelementptr inbounds i8, ptr %0, i64 448
  %bound_check_8bytes30 = load i64, ptr %3, align 8
  %maddr = getelementptr inbounds i8, ptr %2, i64 2437
  store i64 0, ptr %maddr, align 1
  %maddr2 = getelementptr inbounds i8, ptr %2, i64 9865
  store <2 x i64> zeroinitializer, ptr %maddr2, align 1
  %maddr5 = getelementptr inbounds i8, ptr %2, i64 3277
  store i16 0, ptr %maddr5, align 1
  %maddr7 = getelementptr inbounds i8, ptr %2, i64 2908
  store <2 x i64> zeroinitializer, ptr %maddr7, align 1
  %maddr9 = getelementptr inbounds i8, ptr %2, i64 1155
  store <2 x i64> zeroinitializer, ptr %maddr9, align 1
  %cmp = icmp ult i64 %bound_check_8bytes30, 42463
  %maddr19 = getelementptr inbounds i8, ptr %2, i64 3254
  %maddr22 = getelementptr inbounds i8, ptr %2, i64 4332
  br label %block2_end

block2_end:                                       ; preds = %block0_begin, %block1_end
  br i1 %cmp, label %got_exception, label %block1_end

block1_end:                                       ; preds = %block2_end
  store i16 0, ptr %maddr19, align 1
  store i64 0, ptr %maddr22, align 1
  br label %block2_end

got_exception:                                    ; preds = %block2_end
  tail call void @aot_set_exception_with_id(ptr %0, i32 2)
  ret <2 x i64> zeroinitializer
}

attributes #0 = { noinline }

```

you can see that your main function(aot_func_internal#1) is an endless loop.

teyahb8, 2025-08-06:
Thanks for confirming with the nice explanation.