wasm-micro-runtime/wasm-micro-runtime #4640: Union { 0 } initializer is no longer clean in GCC 15

XeniaLu, 2025-09-21:
GCC: https://gcc.gnu.org/gcc-15/changes.html:

> `{0}` initializer in C or C++ for unions no longer guarantees clearing of the whole union (except for static storage duration initialization), it just initializes the first union member to zero. If initialization of the whole union including padding bits is desirable, use `{}` (valid in C23 or C++) or use `-fzero-init-padding-bits=unions` option to restore old GCC behavior.

This is critical for `WASMValue` whose first member is `int32_t`, but obviously it has larger members.

Discussions: https://news.ycombinator.com/item?id=43792948

yamt, 2025-09-22:
> I think if this were true, we do need to add -fzero-init-padding-bits=unions for newer versions of GCC.

as mentioned in the referenced discussion, it isn't only gcc.
clang didn't have the extension until recently. https://github.com/llvm/llvm-project/pull/97121
i suspect it's simpler for us to use memset.

yamt, 2025-09-24:
> I will check the union {} initializer. I think we can replace them if there are not too many

is it ok for us to require c23?

lum1n0us, 2025-09-24:
If embedded platforms' compilers are compatible with C23.